FrameOS/frameos #14: Unknown waveshare driver variant epd2in13[.*]

sowbug, 2024-01-03:
Trying with this board: https://www.aliexpress.us/item/2251832623543336.html. Picked every variant of the 2.13" 250x122 Waveshare device.

```
2024-01-03 23:16:20
- Generating app: 351b3071-4b50-4471-b0f7-a94b4d94ba56 (unsplash)
2024-01-03 23:16:20
Unknown waveshare driver variant epd2in13

[...]

2024-01-03 23:17:24
- Generating app: 351b3071-4b50-4471-b0f7-a94b4d94ba56 (unsplash)
2024-01-03 23:17:24
Unknown waveshare driver variant epd2in13g

[...]

- Generating app: 351b3071-4b50-4471-b0f7-a94b4d94ba56 (unsplash)
2024-01-03 23:17:39
Unknown waveshare driver variant epd2in13_V2

[...]

- Generating app: 351b3071-4b50-4471-b0f7-a94b4d94ba56 (unsplash)
2024-01-03 23:17:50
Unknown waveshare driver variant epd2in13_V4
```

The fact that it doesn't know about the "driver variant" suggests the problem is not a mismatch with the hardware, but rather a problem with the software toolchain. But that's just speculation.

mariusandra, 2024-01-03:
Indeed, this is a bug. More specifically, I rushed out the implementation with just two waveshare drivers, and left adding the remaining ones as one of the [next todo items](https://github.com/FrameOS/frameos/issues/1).

This blew up on HN faster than I thought, so it's not done yet. Essentially [this folder](https://github.com/FrameOS/frameos/tree/main/frameos/src/drivers/waveshare/ePaper) needs a few extra files for all the drivers to work. The readme gives hints on how to do that. Feel free to have a look, or I'll get to it first chance I get, but not before tomorrow.

sowbug, 2024-01-03:
Good luck with the project! Was looking for something turnkey.

closed 2024-01-06 by mariusandra

mariusandra, 2024-01-06:
**All 2-color and 3-color waveshare displays should now work.**

I've added all the waveshare drivers I found under https://github.com/waveshareteam/e-Paper/tree/master/RaspberryPi_JetsonNano/c/lib

These are all the ones with the SPI interface on the list on [any of their store pages](https://www.waveshare.com/product/displays/e-paper/4.37inch-e-paper-g.htm). 

[image]

Everything except these ones with "Parallel port" is now wired up.

[image]

The only problem is that I didn't implement any color reduction algorithms for the 4-color, 4-grayscale and 7-color displays yet. I sadly don't have any frames to test with. Thus this goes on the TODO list.

I do however also have one 2.13" Waveshare display, and got it working with the V3 driver.