FrameOS/frameos #6: Generated SSH keypair not working

ikornaselur, 2024-01-03:
Struggled a little bit setting FrameOS up.

For reference, I have a Pi Zero set up with the ~5 inch Pimoroni inky screen. I've deployed the controller on a separate server and trying to set the connection up.

I do not have passwords enabled for SSH, so I was happy to see that SSH keypairs are supported in settings. Generating a keypair and adding the public key to `.ssh/authorized_keys` on the Pi Zero fails to connect because the key doesn't seem to be a valid RSA key

```
2024-01-03 17:22:09 Deploying frame <name> with build id mkjdapkjexeb
2024-01-03 17:22:09 Connecting via SSH to <user>@192.168.86.45 (keypair)
2024-01-03 17:22:09 not a valid RSA private key file
``` 

I tried creating my own SSH keys (using ed25519) and saving that directly in settings, which gives me a different error

```
2024-01-03 17:21:02 Deploying frame <name> with build id tqfnsxmojkfk
2024-01-03 17:21:02 Connecting via SSH to <user>@192.168.86.45 (keypair)
2024-01-03 17:21:02 unpack requires a buffer of 4 bytes
```

I then finally created a RSA keypair with `ssh-keygen -t rsa` and was able to get that working. If the generated SSH keypairs work on your end, then I'm happy to provide any further information for troubleshooting.

mariusandra, 2024-01-03:
Hey @ikornaselur , thanks for trying out FrameOS and for reporting back!

Indeed, currently only [RSA keys are supported](https://github.com/FrameOS/frameos/blob/main/backend/app/utils/ssh_utils.py#L51). It doesn't seem too hard to add `ed25519` though. I'll see what can't be done, but can't promise exactly when I'll get to it.

ikornaselur, 2024-01-03:
No worries! I think it's fair that only RSA keys are supported. But the main issue is still that the generated keypair from the UI, with this button:

[image]

produces the "not a valid RSA private key file" error, so seemingly the key being generated by FrameOS is invalid. I was only able to get a manually generated keypair working.

Does a generated keypair from the UI work on your end? I'm happy to investigate further on my end and see if I can identify the issue.

Oh, and for further reference, I am deploying the controller with docker and the `:latest` tag

sowbug, 2024-01-03:
Same here. Also docker with latest tag.

mariusandra, 2024-01-04:
This should work now!

I tried changing the default key from RSA to ed25519, but ran into some weird issue. The generated key seemed much too short, even for ed25519, and ChatGPT seems to think it's better to defer to `ssh-keygen` than use python with ed25519. So for another day.

closed 2024-01-04 by mariusandra

alexrudd2, 2024-01-17:
Strange, I thought this was fixed by #27  but it recurred with me again, running ac2e102795db3b6b07704681b36308d18d8195b5.

Working around for now since I want to (1) test #30  again and (2) try out the dithering on a red/black/white InkyWhat.  (I had code lying around locally that tried to dither with `Pillow` but never quite worked)