FrameOS/frameos #7: No such file: nimassets.nim

sowbug, 2024-01-03:
I'm following the setup instructions. I ran into the issue that the controller's generated RSA keys don't work, but I manually worked around it. When I tried to deploy to my RPi 3 Model B with a Waveshare 7.5 display, I got this at the end of the logs:

```
2024-01-03 20:33:34
[1m/tmp/nimble_56/githubcom_xmonadernimassets_0.2.4/src/nimassets.nim(3, 8) [0m[33mWarning: [0muse the nimble packages `malebolgia`, `taskpools` or `weave` instead; threadpool is deprecated[36m [Deprecated][0m[0m
2024-01-03 20:33:35
The command exited with status 1
2024-01-03 20:33:35
[Errno 2] No such file or directory: '/tmp/tmpsy0pgjpp/frameos/\x1b[1m/tmp/nimble_56/githubcom_xmonadernimassets_0.2.4/src/nimassets.nim'
```

alexrudd2, 2024-01-03:
What do you get when you run `uname -a` on the Raspberry Pi 3B?

I _think_ it will be `armv71`.

Here's a suggested change to the architecture detection code in `frameos/backend/apps/tasks/deploy_frame.py`
```python
-- if arch == "aarch64":
++ if arch == "aarch64" or arch == 'armv71`
    cpu = "arm64"
elif arch == "armv6l":
    cpu = "arm"
elif arch == "i386":
    cpu = "i386"
else:
    cpu = "x86_64"
```

sowbug, 2024-01-03:
I reimaged with the 64-bit lite OS choice (which I now see the instructions suggested), and `uname -m` now reports `aarch64`. I'm still waiting for the setup to complete, but I bet this was the problem.

Unless there's a reason why the other architectures can't work, it would be really nice to be more flexible about them (as I see @alexrudd2 has already done!).

closed 2024-01-03 by mariusandra

sowbug, 2024-01-03:
Aw, heck, and now I just did this:

[image]

mariusandra, 2024-01-03:
Ouch, that hurts :(.

PR #9 (2024-01-03): Fix architecture detection code
Fix `amd64` and  `armv7l` (RPi 3 / 3+) architectures.  Closes #7
commit 0daf6eae (2024-01-03): Fix armv7l (RPi 3 / 3+ architecture).  Closes #7