FrameOS/frameos #8: Frame stuck at `uninitialized`

alexrudd2, 2024-01-03:
Thanks for this project! (coming from HN)﻿

I've been unable to get a frame to initialize.  After adding a frame in the web UI it says `uninitialized` forever, and I don't see any relevant logs to suggest what's wrong.  The Pi `sshd` logs don't show any attempt to connect to the frame.

This occurs with either the docker image or running manually.
The RPi4 running Raspbian 64bit (bookworm) is accessible on `blackberrypi.local`
My Linux Mint laptop is accessible on `antorak.local`.
They can ping each other fine, and SSH with keyfile works.
I have added the FrameOS keyfile to `~/.ssh/authorized_users` on the Pi. 

```
// GET frames returns status 200
{
  "frames": [
    {
      "background_color": "white",
      "color": null,
      "device": "web_only",
      "frame_host": "blackberrypi.local",
      "frame_port": 8787,
      "height": null,
      "id": 1,
      "interval": 300.0,
      "metrics_interval": 60.0,
      "name": "Inky",
      "rotate": 0,
      "scaling_mode": "cover",
      "scenes": [
        {
          "edges": [
            {
              "id": "d8ea5451-6154-4f91-9416-c4bf75f90d1e",
              "source": "65ece135-83a4-4103-a9ae-d83dec627e90",
              "sourceHandle": "next",
              "target": "db5f595d-2b15-4506-b7c1-d03bbf5e0e37",
              "targetHandle": "prev"
            }
          ],
          "id": "default",
          "nodes": [
            {
              "data": {
                "keyword": "render"
              },
              "height": 72,
              "id": "65ece135-83a4-4103-a9ae-d83dec627e90",
              "position": {
                "x": 259.18108974358967,
                "y": 379.3192307692308
              },
              "type": "event",
              "width": 132
            },
            {
              "data": {
                "config": {},
                "keyword": "unsplash"
              },
              "height": 102,
              "id": "db5f595d-2b15-4506-b7c1-d03bbf5e0e37",
              "position": {
                "x": 598.6810897435896,
                "y": 412.8192307692308
              },
              "type": "app",
              "width": 133
            }
          ]
        }
      ],
      "server_api_key": "92de851735332deb545b5471ba7d0061bce528bdb13efef6bdae9a49a8754132",
      "server_host": "antorak.local",
      "server_port": 8989,
      "ssh_pass": null,
      "ssh_port": 22,
      "ssh_user": "pi",
      "status": "uninitialized",
      "version": null,
      "width": null
    }
  ]
}
```

```
huey_consumer.py app.huey.huey --worker-type=greenlet --workers=10 --flush-locks
[2024-01-03 14:31:58,518] INFO:huey.consumer:MainThread:Huey consumer started with 10 greenlet, PID 225691 at 2024-01-03 20:31:58.518966
[2024-01-03 14:31:58,519] INFO:huey.consumer:MainThread:Scheduler runs every 1 second(s).
[2024-01-03 14:31:58,519] INFO:huey.consumer:MainThread:Periodic tasks are enabled.
[2024-01-03 14:31:58,519] INFO:huey.consumer:MainThread:The following commands are available:
+ app.tasks.deploy_frame.deploy_frame
+ app.tasks.reset_frame.reset_frame
+ app.tasks.restart_frame.restart_frame
```

```
(env) alex@antorak:~/git/frameos/backend$ python3 run.py
Starting server
 * Restarting with stat
Starting server
 * Debugger is active!
 * Debugger PIN: 354-832-218
```

alexrudd2, 2024-01-03:
Hahaha, you have to click the frame name to pull it up.

 d'oh

mariusandra, 2024-01-03:
Haha, will leave it open to make the UX smoother here.

mariusandra, 2024-01-04:
Now after adding a frame, we redirect you to the frame's screne directly.

closed 2024-01-04 by mariusandra