FrameOS/frameos #24: Not all Pimoroni inky displays support saturation kwarg

alexrudd2, 2024-01-04:
The Python code calls `inky` with a kwarg `saturation`
https://github.com/FrameOS/frameos/blob/09bc0fda0379d74ed67e1433169064600a1dd388/frameos/vendor/inkyPython/run.py#L44

This is not always possible.
See the various drivers in ﻿https://github.com/pimoroni/inky/tree/main/library/inky.  Not all of them support `saturation` as a keyword arg, and the ones that don't aren't e.g. accepting `**kwargs` as a catch-all.

In particular, the Inky wHAT (Red/Black/White) only works if the argument is deleted.
Smarter detection of the inky model is necessary, I think.

mariusandra, 2024-01-04:
Thanks for pointing this out! Will improve this later today, unless your PR is faster 😅

closed 2024-01-04 by mariusandra

alexrudd2, 2024-01-05:
> Thanks for pointing this out! Will improve this later today, unless your PR is faster 😅

Hahaha, I had my first day of a "real job" today, sorry :P 

Anyways, the change worked for me!