FrameOS/frameos #30: Sporadic "Nim executable not found"

alexrudd2, 2024-01-05:
```
2024-01-05 06:16:56
Deploying frame Inky with build id bjzqmketgrsj
2024-01-05 06:16:56
Nim executable not found
```

`Save & Redeploy` a frame does not always work.  Sometimes it works first try, sometimes it takes repeated clicking.  I have not been able to determine a pattern.

```
alex@antorak:~$ cat /etc/os-release
NAME="Linux Mint"
VERSION="21.2 (Victoria)"
ID=linuxmint
ID_LIKE="ubuntu debian"
VERSION_ID="21.2"
VERSION_CODENAME=victoria
UBUNTU_CODENAME=jammy

alex@antorak:~$ which nim
/home/alex/.nimble/bin/nim

alex@antorak:~$ nim --version
Nim Compiler Version 2.0.2 [Linux: amd64]
```

mariusandra, 2024-01-14:
Are you running it via docker or locally? Did you have any insights in the meanwhile why this could be happening?

This should ideally _just work_ with docker. Locally, on mac OS, I haven't experienced such an error. Happy to do what I can to help debug it 🤔

alexrudd2, 2024-01-16:
I've been running locally (on Linux Mint).

I want to get you some more information soon, but haven't had a chance to
try again.

alexrudd2, 2024-01-17:
[image]

The very first time I ran the docker image.  Every subsequent time I ran manually with `honcho start`.... not realizing the docker container kept running... and running... even after a computer restart.

I bet this explains so many oddities I was seeing!

closed 2024-01-17 by alexrudd2