FrameOS/frameos #35: Error: cannot open file: pixie

lvlts, 2024-01-08:
I've prepared a RPi Zero W as instructed by the documentation, with a HyperPixel 2.1 Round display (the Linux console works on the round display).

When deploying, I get the following error:

```
2024-01-08 10:11:02 Deploying frame Frame 1 with build id qsgcfayzctos
2024-01-08 10:11:02 Connecting via SSH to root@frame1.local (keypair)
2024-01-08 10:11:03 Connected via SSH to root@frame1.local
2024-01-08 10:11:03 - Getting target architecture
2024-01-08 10:11:03 > uname -m
2024-01-08 10:11:05 armv6l
2024-01-08 10:11:05 - Copying build folders
2024-01-08 10:11:06 - Applying local modifications
2024-01-08 10:11:06 - Generating scene: default
2024-01-08 10:11:06 - Generating app: a5c68f1a-e3c2-452e-b4f1-85ebc3ad703a (unsplash)
2024-01-08 10:11:06 - Creating build archive
2024-01-08 10:11:06 - No cross compilation. Generating source code for compilation on frame.
2024-01-08 10:11:06 $ cd /tmp/tmpw7z8cfa_/frameos && nimble assets -y && nim compile --os:linux --cpu:arm --compileOnly --genScript --nimcache:/tmp/tmpw7z8cfa_/build_qsgcfayzctos src/frameos.nim 2>&1
2024-01-08 10:11:07 ..................................................................................................................................................
2024-01-08 10:11:07 /tmp/tmpw7z8cfa_/frameos/src/frameos/frameos.nim(1, 29) Error: cannot open file: pixie
2024-01-08 10:11:07 The command exited with status 1
2024-01-08 10:11:07 'NoneType' object has no attribute 'split'
2024-01-08 10:11:07 SSH connection closed
```

Is there anything I can do to include the pixie nimble package in the archive?

mariusandra, 2024-01-08:
Hey @lvlts , thanks for trying FrameOS! Can you try again with the latest docker image? I believe I [fixed it](https://github.com/FrameOS/frameos/commit/ebbdf0c7a5860e5fa6e7bc1411fe5a0bcc37ac3c), but haven't had a chance to personally verify yet.

lvlts, 2024-01-08:
thank you @mariusandra ! it works now. closing.

closed 2024-01-08 by lvlts