FrameOS/frameos #36: Clock overlay time not update

ngothanhnghi, 2024-01-08:
I followed the demo instruction and add the clock overlay as follow:

Event rendering --> Vanilla boilerplate "Code" --> Clock overlay

Actual behavior
The vanilla log message was printed with a period, however, the clock time rendered in the e-ink panel isn't updated. 

Expected behavior:
Time is updated every now and then when rendering event happens.

mariusandra, 2024-01-08:
Hey @ngothanhnghi , can you share some screenshots to help me reproduce the situation? From the above I'm sadly not able to provide much help.

What have you tried? Does it work if you change the order of the apps?

ngothanhnghi, 2024-01-08:
This is the setup
[image]

Log 
[Uploading overlay_clock.log…]()

ngothanhnghi, 2024-01-08:
The code part is added to print out the log line "Hello from render "

mariusandra, 2024-01-09:
Under "Frame Settings", what do you have as the refresh interval?

[image]

ngothanhnghi, 2024-01-09:
thank you that fixes the problem, not aware of this setting. should mention it in the guide/demo as this is an important feature

closed 2024-01-09 by ngothanhnghi