FrameOS/frameos #48: Fails to find/compile pixie

alexrudd2, 2024-01-17:
ac2e102795db3b6b07704681b36308d18d8195b5 on Linux (no docker)

```
CONTROLLER

>nim --version
Nim Compiler Version 2.0.2 [Linux: amd64]
Compiled at 2023-12-15
Copyright (c) 2006-2023 by Andreas Rumpf

git hash: c4c44d10df8a14204a75c34e499def200589cb7c
active boot switches: -d:release

>nimble list --installed

asynctools  [(version: 0.1.1, checksum: 54314dceabb06b20908ecb0f2c007e9ff3aaa054)]
bumpy  [(version: 1.1.2, checksum: acc789c4e31b3ee969a13dd0c238d808f39b19c7)]
chroma  [(version: 0.2.7, checksum: f873f4b97e87a82e27e0b85bf9130fe3e320a2b8)]
crunchy  [(version: 0.1.10, checksum: 49e365a8a558293137dc333f25337be4fd033b37), (version: 0.1.9, checksum: 3afda170fbe40158cd78e00375e34690f1fae263)]
flatty  [(version: 0.3.4, checksum: 5775e6ea6ca339efc5bd37b082b8294342d49dc5)]
httpbeast  [(version: 0.4.2, checksum: ceacc8345c9513bd69c4bdb36dd2cb270e8614c4), (version: 0.4.1, checksum: b23e57a401057dcb9b7fae1fb8279a6a2ce1d0b8)]
jester  [(version: 0.6.0, checksum: a5120cfff525e23be29bb2ff47b6ba2f4b2674a3), (version: 0.6.0, checksum: 4834f85e61ae39f6b6acfb74d3bbba62d8779b66)]
linuxfb  [(version: 0.1.0, checksum: 8e639fafa952f3e9d0315f181aa05e0694603bfc)]
nimassets  [(version: 0.2.4, checksum: 71d5510ad86a323fc0ad5dc6b774261e80fe0361)]
nimsimd  [(version: 1.2.6, checksum: a371cce59834c8d56091f5c891fd182e2eb9b3c4)]
pixie  [(version: 5.0.6, checksum: fd4c5d12bba41aac760fc86c2de820e1a367dbfd)]
psutil  [(version: 0.6.1, checksum: 137403c200c58c6cd566ab5fbfd597b535303fc6)]
unpack  [(version: 0.4.0, checksum: d65465ca8e74e9fcb590255491c8fe7763e38c42)]
vmath  [(version: 2.0.0, checksum: c90a664f3425d26219e1262ed9a888b859b3127c)]
wiringPiNim  [(version: 0.1.0, checksum: 88e1225a806aae76fd4de1d08570c44986a30f00)]
ws  [(version: 0.5.0, checksum: ae4daf4ae302d0431f3c2d385ae9d2fe767a3246)]
zippy  [(version: 0.10.11, checksum: 12e761e18ccc0a1b0e931c74dd02bb4b74544d56)]
zstd  [(version: 0.9.0, checksum: 20b23158e94f01ea0c4bf419a21b0feabe70bf31)]

> printenv | grep PATH

PATH=/home/alex/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/alex/.local/bin:/home/alex/.local/bin:/home/alex/.nimble/bin

> which nimble

/home/alex/.nimble/bin/nimble

```
```
FRAME

2024-01-17 02:07:19
Hint: used config file '/tmp/tmpxp4v2no8/frameos/nim.cfg' [Conf]
2024-01-17 02:07:19
Hint: used config file '/tmp/tmpxp4v2no8/frameos/config.nims' [Conf]
2024-01-17 02:07:19
..................................................................................................................................................
2024-01-17 02:07:19
/tmp/tmpxp4v2no8/frameos/src/frameos/frameos.nim(1, 29) Error: cannot open file: pixie
2024-01-17 02:07:19
The command exited with status 1
2024-01-17 02:07:19
Error in src/frameos/frameos.nim:1:29
2024-01-17 02:07:19
Line 1: import json, asyncdispatch, pixie, chroma, strutils
2024-01-17 02:07:19
....................................^
2024-01-17 02:07:19
Failed to generate frameos sources
2024-01-17 02:07:19
SSH connection closed
```

mariusandra, 2024-01-17:
Ah, I know what this is about: you also need to run `nimble setup` in the `frameos` folder.

I've updated the [contribution guide](https://github.com/FrameOS/frameos/commit/77271f4c3741b0be383199584bb5ad02980388ff) accordingly.

closed 2024-01-17 by alexrudd2

mariusandra, 2024-01-17:
Ah, thanks. Fixed!