FrameOS/frameos #87: Seriously struggling to figure out making my own scenes?

Hexagram6716, 2024-05-18:
I have tried and tried again and I cant even get a simple text example

I have never used node stuff before a video tutorial of showing home assistant sensors. etc would be a god send

The project looks exactly perfect for what I want but the lack of explanation for people with low level understanding of the interface

mariusandra, 2024-05-21:
Hey @Hexagram6716 , I understand your frustrations. This project is an early-stage one-man side project, hence things are still rough around the edges. I've been working through [the tasklist](https://github.com/FrameOS/frameos/issues/1), focusing on the larger backend tasks that are needed to make any of this possible. I have a few more items left before my focus will switch to building a great UI and a great experience, which includes documentation and video guides. So I ask for just a little more patience :). At present the best way to learn to compose scenes is to edit the existing samples, and learn from there.

mariusandra, 2024-07-14:
Hey @Hexagram6716 , I have finally updated the documentation.

Since the last comments here, I added a few features that make a huge difference in how scenes are built. Had I documented the old system, I would have needed to rewrite most of it now, and definitely record all new screencasts. I only have limited time I can spend on this project, and since writing docs for a dead system would not be very motivating, I kept improving the system instead. I now reached a natural stopping point and finally cleaned up the docs: https://frameos.net/

I'm very curious to hear if the new docs improve things, and what's still missing. A longer youtube video is coming soon as well.

closed 2024-07-22 by mariusandra

mariusandra, 2024-11-09:
Hey @Hexagram6716 , I finally have a video out explaining the system: https://frameos.net/blog/nimconf-2024 - In case you're still interested in this project, that might help understand the interface a bit better.