FrameOS/frameos #89: Folder gallery

WizardUli, 2024-05-20:
It would be nice to have a template for just displaying photos from a folder in a selectable order (sorted alphabetically by path or random).
I want to use FrameOS just for displaying some photos on a (mostly) offline device from a folder on a flash drive (not some cloud) but haven't had time to learn Nim just yet to write my own app.

I imagine that exact use case is probably very common (e.g. gifting grandma who does not have internet a frame preloaded with family pictures changing once a day) hence I'm suggesting adding it as a template.

mariusandra, 2024-05-20:
Hey, that's not a bad idea! I've added a "local image" app with the [latest updates]https://github.com/FrameOS/frameos/pull/88), and a corresponding sample to the repository. 

If you intend this frame to run forever, I do at present also suggest enabling nightly restarts (under the frame's settings), as there seems to be a small memory leak with the logger... :/.

WizardUli, 2024-05-21:
Seems to work nicely. Thanks for the quick response.

Maybe the counter could be replaced with a filename (or rather a path in case we want sub-folders) and let it be publicly a changeable state in case we want to switch to a specific image manually or on some event. But that can wait until I learn Nim and then I can to it myself. ( I have always want to learn Nim anyway since it compiles to C so I can play with esoteric non-ARM non-RISC-V MCUs with their own esoteric C toolchains :D )