FrameOS/frameos #95: https://source.unsplash.com/random seems down

coro1404, 2024-06-08:
https://source.unsplash.com/random (which is used for the random image integration) seems to be down (for good?)

[image]

mariusandra, 2024-06-22:
Indeed, it appears we can't use this anymore.

I've [added a new app](https://github.com/FrameOS/frameos/pull/96) that uses the official unsplash API. The downside is you'll need to register and get an API token. The upside is better cropping and no more "one word keyword" limit.

You'll most likely need to upgrade FrameOS to use it. I released a [large change](https://github.com/FrameOS/frameos/pull/93) to the way apps are built and the new Unsplash app takes advantage of those changes.

mariusandra, 2024-06-23:
Going to close this as it works for me with the new app. You do need an API key, but the "demo" mode gives you 50 requests per hour, which should be plenty for most casual usecases.

closed 2024-06-23 by mariusandra