FrameOS/frameos #105: App for Download Random Pic From PhotoPrism

L0rz, 2024-08-17:
Hello There,

I use the power of OpenAi and my little skills to create an app to use the API from photoprism Link: https://www.photoprism.app
and give back a Random link of a Photo to use in frameOS.

I like to share my code but have no clue how so i opend an issue here.

the code in app.nim:
```
import options
import lib/httpclient
import frameos/apps
import frameos/types
import json
import strutils

type
  AppConfig* = object
    url*: string
    albumQuery*: string
    apiToken*: string  # Der API-Token für die Authentifizierung

  App* = ref object of AppRoot
    appConfig*: AppConfig
    downloadToken*: string  # Speichert den downloadToken

proc getSessionDownloadToken(self: App): string =
  let sessionUrl = self.appConfig.url & "/api/v1/session"
  let client = newHttpClient(timeout = 30000)
  client.headers.add("Authorization", "Bearer " & self.appConfig.apiToken)

  try:
    let response = client.getContent(sessionUrl)
    let jsonResponse = parseJson(response)
    
    if jsonResponse.hasKey("config") and jsonResponse["config"].hasKey("downloadToken"):
      self.downloadToken = jsonResponse["config"]["downloadToken"].getStr()
      return self.downloadToken
    else:
      self.logError "Kein downloadToken in der Antwort gefunden."
      return ""
  except CatchableError as e:
    self.logError "Fehler bei der Anfrage: " & e.msg
    return ""
  finally:
    client.close()

proc getRandomPhotoHash(self: App): string =
  let photosUrl = self.appConfig.url & "/api/v1/photos?count=1&offset=0&q=" & self.appConfig.albumQuery & "&order=random"
  let client = newHttpClient(timeout = 30000)
  client.headers.add("Authorization", "Bearer " & self.appConfig.apiToken)
  
  try:
    let response = client.getContent(photosUrl)
    let jsonResponse = parseJson(response)
    
    if jsonResponse.kind != JArray or jsonResponse.len == 0:
      self.logError "Kein Bild gefunden."
      return "Kein Bild gefunden."
    
    let photoHash = jsonResponse[0]["Hash"].getStr()
    return photoHash
  except CatchableError as e:
    self.logError e.msg
    return e.msg
  finally:
    client.close()

proc getPhotoDownloadUrl(self: App, photoHash: string): string =
  # Erstelle die Download-URL basierend auf dem Hash und dem downloadToken
  return self.appConfig.url & "/api/v1/t/" & photoHash & "/" & self.downloadToken & "/fit_1280"

proc get*(self: App, context: var ExecutionContext): string =
  if self.getSessionDownloadToken().len == 0:
    return "Fehler beim Abrufen des downloadToken."
  
  let photoHash = self.getRandomPhotoHash()
  
  if photoHash != "Kein Bild gefunden.":
    return self.getPhotoDownloadUrl(photoHash)
  else:
    return photoHash

```
code in config.json
```
{
  "name": "Download URL",
  "description": "Download a URL into a string",
  "category": "data",
  "version": "1.0.0",
  "fields": [
    {
      "name": "url",
      "type": "text",
      "value": "",
      "required": true,
      "label": "URL",
      "placeholder": "https://domain/content"
    },
    {
      "name": "albumQuery",
      "type": "text",
      "value": "",
      "required": true,
      "label": "albumQuery",
      "placeholder": "album:albumID"
    },
    {
      "name": "apiToken",
      "type": "text",
      "value": "",
      "required": true,
      "label": "apiToken",
      "placeholder": "123456789-1234567-12345689-12315686"
    }
  ],
  "output": [
    {
      "name": "result",
      "type": "string"
    }
  ],
  "cache": {
    "enabled": true,
    "inputEnabled": true,
    "durationEnabled": true,
    "duration": "900"
  }
}

```

[image]

UrbanCircles, 2024-11-19:
awesome - is there an integration with Immich albums?

L0rz, 2024-12-03:
Hi, The Immich api remove the support of getting a random picture: https://immich.app/docs/api/get-random
so there is a little bit more to do,