FrameOS/frameos #108: More detailed step-by-step tutorials for Inky frame

UrbanCircles, 2024-11-19:
Saw the high-level outline of how to get this working, but that wasn't enough for a noob like me (which is probably a significant part of the audience).

Any chance you can put together step by step tutorials to get this up and going?

disruptivepatternmaterial, 2024-12-20:
Which frame and hardware specifically? I have Inky Impression 7.3" and got it up and running in minutes.

There is one BIG thing you need to watch out for, if you have anywhere near the more recent version of the OS on your RPi, the install scripts have to be changed significantly. The good news is people did the work for you!

The instructions here work: https://github.com/pimoroni/inky they create the venv needed for the setup.

Once you have that done, you point frameos at it and do the initial install via the web browser (frameos) like it says in the docs.

Ask some more detailed or specific questions and I am happy to answer...I have 5 of them working and it was super slick!

mariusandra, 2024-12-21:
Hey all! I've been working towards making FrameOS much easier to use, and will soon release a much better tutorial/video about building a frame and getting everything set up. It'll be on the basis of the 12.48" waveshare panel, but 95% of the content should apply no matter what you use. The inky frames do require a manual "install drivers" step, as also mentioned in [their respective guides](https://frameos.net/devices/pimoroni-inky-impression-7.3inch-eink) (happy to get a PR if something changed 😅).

One big part of this "making it easier" project is to release FrameOS as a [Home Assistant Addon](https://github.com/FrameOS/frameos-home-assistant-addon) (in addition to also working standalone). I've been hard at work on this for some time now. It works, but it won't show up in the Home Assistant sidebar just yet. I'm in the progress of [a large migration](https://github.com/FrameOS/frameos/pull/109) to make that possible and solve some other technical hurdles.

When that works, my next step will be to actually get the frames into home assistant, so you could send commands (as part of automations), see the image, etc. I expect all this will be done by the end of next week, after which I'll work on a new getting started guide/video, and then continue generally improving the platform. I won't release a new guide now, as I'll have to redo it in a week, so just a bit more patience please :).

UrbanCircles, 2024-12-21:
This is fantastic news!! Might even pick one of these frames up, just for this purpose 😂

mariusandra, 2024-12-29:
For those following along, the Home Assistant addon is now ready: https://frameos.net/guide/backend/#home-assistant-addon

You can now run FrameOS within Home Assistant's sidebar, even though it won't create native devices for the frames just yet. That will be for later. I'll work on various low hanging improvements and on a nicer getting started guide next.