FrameOS/frameos #110: not an issue but not sure where to have a discussion :)

disruptivepatternmaterial, 2024-12-20:
This project is really cool! I was about to start to write a way jankier version and then found it.

I have 6  eink frames embedded in the wall of my bathroom to do rotating art and into in the morning. I have a larger repurposed monitor that I also show images on from trips and family and such.

I want them all to be identical. What would you do as a best practice to make that happen? Right now I have a basic SD RAM example working and I had to remake it 8 times. I would love to understand how you might just deploy the same to all devices.

Second is that I use Home Assistant and for the life of me I cannot figure out how to get it to show the status of a sensor using your flow diagram. Anyone have an example of a flow that might make that happen? I tried modifying the text message one, but no luck.

mariusandra, 2024-12-21:
This is a great place for discussions, and thanks for the kind words!

I assumed one day I'll need to support this "multi device deploy" feature, but that hasn't really been on my list just yet, as all devices I have around the house have slightly different things running on them. 

The best way to do this today is to create one scene, save it as a template ("save to my scenes"), install that on other devices, and then deploy. 

[image]

It's not the end result, but should be fine if you don't make changes often... Yet if you do... I'd be happy to make it work. I have an idea on how to do it, however as I wrote [here](https://github.com/FrameOS/frameos/issues/108#issuecomment-2558059301), I already have a bunch of work queued up over the holidays that I need to get through first. So.. soon... 😆.

mariusandra, 2024-12-21:
I forgot to reply to this part

> Second is that I use Home Assistant and for the life of me I cannot figure out how to get it to show the status of a sensor using your flow diagram. Anyone have an example of a flow that might make that happen? I tried modifying the text message one, but no luck.

It's a bit complicated right now. Here's the gist of what I have:

[image]

In the first step a "home assistant sensor" plugin makes a HTTP request to the Home Assistant API (set the key in settings), once per second, fetching the entity "water_heater.hot_water" into the key "water_heater" in the scene's state.

Then I run this through a custom app, which calculates the remaining time and stores it under the state key "heatTimer". The app's source looks like this:

```nim
import json, strformat, times
import pixie
import frameos/apps
import frameos/types

type
  AppConfig* = object
    keyword*: string

  App* = ref object of AppRoot
    appConfig*: AppConfig

proc init*(nodeId: NodeId, scene: FrameScene, appConfig: AppConfig): App =
  result = App(
    nodeId: nodeId,
    scene: scene,
    frameConfig: scene.frameConfig,
    appConfig: appConfig,
  )

proc run*(self: App, context: ExecutionContext) =
  if self.scene.state{"water_heater"}{"state"}.getStr == "heat":
    if self.scene.state{"heatStart"}.getFloat == 0.0:
      self.scene.state["heatStart"] = %*(epochTime())
  else: 
    if self.scene.state{"heatStart"}.getFloat != 0.0:
      self.scene.state["heatStart"] = %*(0.0)

  let heatStart = self.scene.state{"heatStart"}.getFloat
  if heatStart == 0.0:
    self.scene.state["heatTimer"] = %*("")
  else:
    let timeRunning = (epochTime() - heatStart)
    let timeLeft = 30.0 * 60.0 - timeRunning
    
    let minutesLeft = (timeLeft / 60.0).int
    let secondsLeft = timeLeft.int mod 60
    let timeLeftString = "" &
      (if minutesLeft < 10: "0" else: "") &
      $minutesLeft & ":" &
      (if secondsLeft < 10: "0" else: "") &
      $secondsLeft
    self.scene.state["heatTimer"] = %*(timeLeftString)
```

The next step is to branch out depending on whether `$state{"water_heater"}{"state"}.getStr` equals `"heat"` or not, and display different types of backgrounds:

[image]

Finally the last step also shows the "heatTimer" calculated above.

Hopefully this helps a bit. If there's anything I could expand on, let me know!

disruptivepatternmaterial, 2024-12-21:
1 - makes sense
2 - ill give it a try!

disruptivepatternmaterial, 2024-12-22:
COOOL I got it working in a basic way - it is showing the temperature over the image.

next question - what is the formatting or basis of $state{"outside_temp"}{"state"}.getStr ? I want to see if I can round the temp and add on the units, and such.

Again, thanks! What a cool project.

disruptivepatternmaterial, 2024-12-22:
for example 

```
"code": "$state{\"outside_temp\"}{\"state\"}.getStr & \"°F\"",
```

how should I round that to one decimal place? is that possible?

mariusandra, 2024-12-22:
That's great!

As for rounding, it's all nim code in the end, so whatever works in nim, should work here. A quick google led me to [this page](https://forum.nim-lang.org/t/7850), which seems to suggest either

```nim
formatFloat(9.779999999999999,format=ffDecimal,precision=2)
```

or

```nim
&"{9.779999999999999:0.2f}"
```

So I'd try something like:

```nim
&"{state{"outside_temp"}{"state"}.getFloat:0.2f}"
```

mariusandra, 2024-12-22:
Oh, and in case it helps, I finally created a Discord for FrameOS: https://discord.gg/pb2EHnfn

I'm happy to help here, especially as it leaves a trail for others to leave, though feel free to join if you need some quick help on things.

I'll link it from the main frameos.net site once I'm done with the HA addon...

disruptivepatternmaterial, 2024-12-24:
in the end we got this to work : formatFloat((try: parseFloat(state{"outside_temp"}{"state"}.getStr) except: 0.0), ffDecimal, 1) & "°F"

THANKS!

closed 2025-01-31 by mariusandra