FrameOS/frameos #130: Pi5 + Inky 7.3 issue

dynstatic, 2025-03-29:
Hello,

I have new Pi5 with a fresh flash of Raspbian Bookworm, apt update && apt dist-upgrade, and I am getting an error that seems to be gpio and pi5 related.

Steps to produce

Pi5 2gb model
Fresh flash of lite 64bit bookworm
as non-root user
sudo apt-get update && sudo apt-get dist-upgrade -y
sudo apt-get install git
git clone https://github.com/pimoroni/inky
cd inky
./install.sh
(Y to all)
reboot
source ~/.virtualenvs/pimoroni/bin/activate
Pimoroni/inky/examples/7color/clear.py
(this works)

Add pub key to .ssh/authorized_keys

Add pi to frameos on ha with inky + button driver
Full deploy

The best I can tell, this is the issue:

2025-03-29 17:07:05
driver:inkybytesReceived=1536150message="rendering on eink display"
2025-03-29 17:07:05
driver:inkyerror="Cannot determine SOC peripheral base address"
2025-03-29 17:07:06
render:driverdevice="pimoroni.inky_impression"ms=497.884

mariusandra, 2025-03-30:
I have an idea what this could be. The latest inky python library is v2.0.0 [(changelog)](https://github.com/pimoroni/inky/blob/main/CHANGELOG.md), however we're using 1.5.0. Perhaps yours is part of a newer batch of screens where it's necessary to be on 2.0.0? 

However after upgrading I now get this

[image]

and I'm not yet sure what that's about. I disabled the gpio button drivers, but no change. Rebooting the frame didn't help.

This was on a pi zero though. Perhaps it's different on a pi 5.

If you feel like manually upgrading to 2.0.0, install it under `/srv/frameos/vendor/inkyPython` (modify `requirements.txt`, then run `source env/bin/activate; pip install -r requirements.txt`). Restart the frameos service and see what happens.

I'll keep digging as well, though I'm going to stop soon for tonight, and pick this up tomorrow.

dynstatic, 2025-03-30:
Updating to 2.0 did fix the issue with the screen. I included the log, it still has a "SPI: Cannot disable chip-select!" warning but does draw. 

From what I was seeing, Pi5 handles GPIO different and older libraries need to be updated to accommodate it.

Thank you for the help!

```
/srv/frameos/vendor/inkyPython $ source env/bin/activate; pip install -r requirements.txt
Looking in indexes: https://piwheels.org/simple, https://pypi.org/simple
Collecting inky==2.0
  Downloading inky-2.0.0-py3-none-any.whl (40 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 40.6/40.6 kB 1.5 MB/s eta 0:00:00
Requirement already satisfied: numpy==1.26.1 in ./env/lib/python3.11/site-packages (from -r requirements.txt (line 12)) (1.26.1)
Requirement already satisfied: pillow==9.5.0 in ./env/lib/python3.11/site-packages (from -r requirements.txt (line 16)) (9.5.0)
Requirement already satisfied: rpi-gpio==0.7.1 in ./env/lib/python3.11/site-packages (from -r requirements.txt (line 18)) (0.7.1)
Requirement already satisfied: smbus2==0.4.3 in ./env/lib/python3.11/site-packages (from -r requirements.txt (line 20)) (0.4.3)
Requirement already satisfied: spidev==3.6 in ./env/lib/python3.11/site-packages (from -r requirements.txt (line 22)) (3.6)
Collecting gpiodevice>=0.0.3
  Downloading gpiodevice-0.0.5-py3-none-any.whl (9.2 kB)
Collecting gpiod
  Using cached gpiod-2.3.0-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (103 kB)
Installing collected packages: gpiod, gpiodevice, inky
  Attempting uninstall: inky
    Found existing installation: inky 1.5.0
    Uninstalling inky-1.5.0:
      Successfully uninstalled inky-1.5.0
Successfully installed gpiod-2.3.0 gpiodevice-0.0.5 inky-2.0.0
```

```
2025-03-30 11:45:20
metricsload=[0.62,0.13,0.04]cpuTemperature=55.65memoryUsage={"total":2109685760,"available":1849212928,"percentage":12.3,"used":175865856,"free":1061158912,"active":277053440}cpuUsage=15.7openFileDescriptors=12
2025-03-30 11:45:21
render:donems=1807.858
2025-03-30 11:45:21
driver:inkyinky=truewidth=800height=480color="multi"
2025-03-30 11:45:21
driver:inkybytesReceived=1536150message="rendering on eink display"
2025-03-30 11:45:21
driver:inkylog="/srv/frameos/vendor/inkyPython/env/lib/python3.11/site-packages/inky/inky_ac073tc1a.py:224: UserWarning: SPI: Cannot disable chip-select!"
2025-03-30 11:45:21
driver:inkylog=" warnings.warn(\"SPI: Cannot disable chip-select!\")"
2025-03-30 11:45:21
driver:inkylog="/srv/frameos/vendor/inkyPython/env/lib/python3.11/site-packages/inky/inky_ac073tc1a.py:285: UserWarning: Busy Wait: Held high. Waiting for 1.00s"
2025-03-30 11:45:21
driver:inkylog=" warnings.warn(\"Busy Wait: Held high. Waiting for {:0.2f}s\".format(timeout))"
2025-03-30 11:45:23
httpget="/image"
```

mariusandra, 2025-03-30:
That's great to hear! I still haven't resumed testing, but one of the thing a lot people had to add [was this line](https://github.com/FrameOS/frameos/pull/131/files#diff-063ea6b5bfa79c74f4bc08ceb7107562200e900aa64d236f86fe464c2ef95f8fR56) to their `/boot/firmware/config.txt` (or `/boot/config.txt`): 

```
dtoverlay=spi0-0cs
```

Perhaps that will get rid of the last warning? I'm not sure.

Also, I probably should've asked earlier: which inky 7.3" do you have? The 2025 version that's now on pre-order (with the 6 color e-ink spectra display) or the older 7-color ACeP one?

dynstatic, 2025-03-30:
Commenting out that line seems to break it. It doesn't seem to render anything also. I'm not sure how to check the screen version.  The python example outputs "Detected 7-Colour 800x480 (AC073TC1A)".

```
2025-03-30 12:15:36
driver:inkyinky=truewidth=800height=480color="multi"
2025-03-30 12:15:36
driver:inkybytesReceived=1536150message="rendering on eink display"
2025-03-30 12:15:36
driver:inkylog="Woah there, some pins we need are in use!"
2025-03-30 12:15:36
driver:inkylog=" ⚠️ Chip Select: (line 8, GPIO8) currently claimed by spi0 CS0"
2025-03-30 12:15:37
metricsload=[0.21,0.06,0.02]cpuTemperature=42.45memoryUsage={"total":2109685760,"available":1875410944,"percentage":11.1,"used":164413440,"free":1750106112,"active":174620672}cpuUsage=31.7openFileDescriptors=12
2025-03-30 12:15:37
render:driverdevice="pimoroni.inky_impression"ms=512.561
2025-03-30 12:15:37
render:sleepms=3599171.691
```

mariusandra, 2025-03-30:
Turns out I couldn't get it working because I had messed up my config.txt from some earlier testing.

I did a clean install and could get my frame to render well with the inky 2.0.0 package. 

https://github.com/FrameOS/frameos/pull/131 - once the tests pass I'll merge this in and we should be using the new driver going forward.

Thanks for reporting and all the helpful debugging!